slapplebags/Tracksoar-Firmware #23: Power Usage

slapplebags, 2015-04-27:
implement software power optimization. Batteries currently only 12 hours at room temp or last 3 hours at -40c.

mkapuscik, 2019-02-27:
So what in particular do you want optimized. Are there specific loops that are blocking that waste power? Sensors that stay on that shouldn't? Unnecessary delay calls? 

It's incredibly helpful to characterize a problem before making an issue about it because it's much easier to tackle a specific set of symptoms that can be veritably solved than to "make it better" which could have no definite end and has infinite scope.

slapplebags, 2019-02-27:
Issue has been resolved, battery life extended to 36 hours at room temp and 12 hours at -40c.

closed 2019-02-27 by slapplebags