slapplebags/Tracksoar-Firmware #29: Integrate RF PA onto Tracksoar board.

mkapuscik, 2015-05-28:
Implement discrete component or monolithic RF PA onto Tracksoar PCB for 144MHz.

mkapuscik, 2015-05-28:
TI's CC11xx series monolithic transceivers have options that support 2-FSK modulation and the [CC1120](http://www.ti.com/lit/ds/symlink/cc1120.pdf) "supports" down to 136.7MHz however is not characterized below 169 MHz.

slapplebags, 2015-05-28:
what about the MAX2606, small package, low external part count, etc.

slapplebags, 2015-06-10:
also an ADL5545 to up the transmit power to 250mw

slapplebags, 2015-07-31:
Mike found a promising new chip, beat the info out of him so we can look into it.

slapplebags, 2015-09-14:
here's an example max2606: http://ram.lijun.li/old/various/fm_xmtr.html

slapplebags, 2016-02-01:
another possiblitity: http://digitalcommons.calpoly.edu/cgi/viewcontent.cgi?article=1156&context=eesp
also max2470 is a whole thing, use the micrl300

mkapuscik, 2019-02-27:
Wrong repo for this

closed 2019-02-27 by mkapuscik